Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant #2: Two unique_ids with name wb1_rated_voltage

SunnyCrockett, 2023-12-13:
Hi Louis

There are two unique_ids with name wb1_rated_voltage. One for Rated voltage and also for Working mode.

Regards Sunny

Louisbertelsmann, 2023-12-13:
Hey,
Read your issues and am gonna fix them soon. Probably some copy paste errors.

Louisbertelsmann, 2023-12-13:
Fixed it. Does it work for you? I unfortunately don‘t have this wallbox and am doing it as an addition to the integration by mkaiser.

SunnyCrockett, 2023-12-14:
Thanks for your quick response!
I have tried it, but still cannot get any data from my wallbox. But it seems to be a different problem now:
`2023-12-14 07:49:57.561 ERROR (SyncWorker_2) [homeassistant.components.modbus.modbus] Pymodbus: SungrowACx: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)`
`2023-12-14 07:50:32.940 DEBUG (SyncWorker_1) [homeassistant.components.modbus.modbus] Pymodbus: SungrowACx: Modbus Error: [Connection] ModbusTcpClient(10.4.89.111:502): Connection unexpectedly closed 2.031667470932007 seconds into read of unbounded read bytes without response from slave before it closed connection`

If you are interested, I would be pleased to give you more information about my hardware setup and what I have tried within my HomeAssistant config.

closed 2024-02-08 by Louisbertelsmann