Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant #3: Modbus reads timing out (SH8.0RT over rear "LAN" Port)

pdressen, 2024-02-01:
Hi,

thank you for creating this integration. It's exactly what I'm currently searching for. Unfortunately I cannot get it to work.
The inverter integration from @mkaiser works flawlessly for me (https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant).

That is until I add just any one modbus item from your intergration to read. Then the whole integration stops working and all sensors become unavailable in Home Assistant. I think this is because nobody responds and the Modbus reads are timing out in the background.
I tried using several device id for the wallbox sensors (eg. 248, 3), but cannot read any register. Also when using a Modbus Tester software or pyModbus, I can read the inverters registers just fine, but no other device id (other than 1) seems to work.

To be clear: I am connecting to the SH8.0RT via the rear integrated LAN port, not the WiNet-S-Dongle. All values from the inverter are working fine. I just want to integrate the wallbox into Home Assistant.
My solar technician has already updated the firmware of the inverter and wallbox, this didn't change anything.

What am I missing here?

Louisbertelsmann, 2024-02-02:
This Integration is currently in development. Unfortunately I don't have enough time to do anything right now.

Louisbertelsmann, 2024-02-08:
@SunnyCrockett just helped to fix the code and it is working now. You just have to use the dongle on the wallbox and look at the slave address in the configuration of it. You might want to look [here](https://github.com/Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant/pull/4).

closed 2024-02-08 by Louisbertelsmann