Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant #6: Charging start/end time is not correct

SunnyCrockett, 2024-02-07:
The values I get for "Charging start time" and "Charging end time" are not correct:
[image]
[image]
I tried to get the correct values with modbus-cli and was successful with the following:
`modbus dongle:502 -s 3 i@21319/I --byte-order=mixed`
`Parsed 0 registers definitions from 1 files`
`21319: 1707007290 0x65bedd3a`
`modbus dongle:502 -s 3 i@21317/I --byte-order=mixed`
`Parsed 0 registers definitions from 1 files`
`21317: 1706937153 0x65bdcb41`

Those values are UNIX timestamps and could be converted to date/time. Unfortunately, I dont know how to configure the "--byte-order=mixed" option in HA yaml. Do you know how to do that?

Thanks & regards
Sunny

Louisbertelsmann, 2024-02-07:
I‘ll figure something out. I did it some time ago, but a lot has changed. I also don‘t have that config anymore.

Louisbertelsmann, 2024-02-07:
You might want to have a look at the swap option from homeassistant. There are a few options, but I don‘t know if the needed one is there.

SunnyCrockett, 2024-02-07:
Thanks for the hint with swap option, I have created a pull request with a fix that works for me:
[image]

Louisbertelsmann, 2024-02-08:
Ok, thanks a lot. I will update the wiki and put it out of beta.

btota, 2024-02-10:
perfect guys! Same for other uint32 values, do we need to add swap:word to all uint32 registers? (ie. "Maximum charging power"|"Total energy")
[image]
[image]

```ruby 
pull request
- name: Maximum charging power
        unique_id: wb1_maximum_charging_power
        slave: !secret wallbox_modbus_slave
        address: 21272 # reg 21273
        input_type: input
        data_type: uint32
        swap: word
        precision: 1
        unit_of_measurement: W
        device_class: Power
        scale: 1
        scan_interval: 60

      - name: Total energy A
        unique_id: wb1_total_energy_a
        slave: !secret wallbox_modbus_slave
        address: 21299 # reg 21300
        input_type: input
        data_type: uint32
        swap: word
        precision: 1
        unit_of_measurement: Wh
        device_class: Energy
        scale: 1
        scan_interval: 600

btota, 2024-02-10:
Working mode is input_type: holding

```ruby
pull request
- name: Working mode
        unique_id: wb1_working_mode
        slave: !secret wallbox_modbus_slave
        address: 21262 # reg 21263
        input_type: holding #input
        data_type: uint16
        scan_interval: 600

SunnyCrockett, 2024-02-10:
I have created a new pull request with some fixes and a new template.

Louisbertelsmann, 2024-02-10:
It doesn't look like that exists, at least not in the official documentation.

btota, 2024-02-10:
Not really - maybe an automation for each scenario is needed, can one also modbusscan with modbus-cli [0-248]?

btota, 2024-02-11:
Wrong datetime (as_local):

```ruby
pull request
template:
  - sensor:
      - name: Charging start time
        unique_id: wb1_charging_start_time
        state: >
          {{ as_local(as_datetime(states.sensor.charging_start_time_raw.state)).strftime("%d.%m.%Y %H:%M") }}

      - name: Charging end time
        unique_id: wb1_charging_end_time
        state: >
          {{ as_local(as_datetime(states.sensor.charging_end_time_raw.state)).strftime("%d.%m.%Y %H:%M") }}

needs to be changed to

template:
  - sensor:
      - name: Charging start time
        unique_id: wb1_charging_start_time
        state: >
          {{ (as_datetime(states.sensor.charging_start_time_raw.state)).strftime("%d.%m.%Y %H:%M") }}
      - name: Charging end time
        unique_id: wb1_charging_end_time
        state: >
          {{ (as_datetime(states.sensor.charging_end_time_raw.state)).strftime("%d.%m.%Y %H:%M") }}

SunnyCrockett, 2024-02-11:
Without the as_local, it is in utc0. I am in utc+1 and wanted it shown like in the iCloudSolar app. Whats your timezone in HA?

btota, 2024-02-11:
Weird. Timezone on HA is gmt+01:00 Berlin.  Using as_local caused right timestamp +1 hour :)

closed 2024-02-23 by SunnyCrockett