Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant #11: Please upload the dashboard files

BigWanna, 2024-02-16:
Hi, without the Dashboard data i can not use your excellent project. Please upload an installation guide how to build the dashboard. Thx a lot!

SunnyCrockett, 2024-02-23:
Hi there
This is mine for now:
```
views:
  - title: Home
    cards:
      - type: entities
        entities:
          - entity: sensor.wb1_dev_number_raw
          - entity: sensor.wallbox_working_mode
          - entity: sensor.mileage_per_kwh
          - entity: sensor.phase_switching_status
          - entity: sensor.total_energy
          - entity: sensor.minimum_charging_power
          - entity: sensor.maximum_charging_power
        title: Infos
      - type: entities
        entities:
          - entity: sensor.charging_energy
          - entity: sensor.charging_duration
          - entity: sensor.charged_range
          - entity: sensor.charging_end_time
          - entity: sensor.charging_start_time
        title: Letztes Laden
      - type: entities
        entities:
          - entity: input_select.set_wb_charger_enable
          - entity: input_select.set_wb_phase_switch
          - entity: input_select.set_wb_remote_control
          - entity: input_number.set_wb1_mileage_per_kwh
          - entity: input_number.set_wb1_output_current
          - entity: automation.wallbox_change_phase_switch_cmd
          - entity: automation.wallbox_charger_enable_cmd
          - entity: automation.wallbox_charger_remote_control_cmd
          - entity: automation.wallbox_output_current
        title: Einstellungen
        show_header_toggle: false
      - type: entities
        entities:
          - entity: sensor.charging_status
          - entity: sensor.charging_power
          - entity: sensor.phase_a_charging_current
          - entity: sensor.phase_a_charging_voltage
          - entity: sensor.phase_b_charging_current
          - entity: sensor.phase_b_charging_voltage
          - entity: sensor.phase_c_charging_current
          - entity: sensor.output_current_setting
        title: Laden aktuell
      - type: gauge
        entity: sensor.charging_power
        needle: true
        min: 4140
        max: 11000
      - type: entities
        entities:
          - entity: sensor.charging_status_raw
          - entity: sensor.charging_start_time_raw
          - entity: sensor.charging_end_time_raw
          - entity: sensor.phase_switching_status_raw
          - entity: sensor.working_mode
          - entity: sensor.power_phases
          - entity: sensor.rated_voltage
          - entity: sensor.remote_control
          - entity: sensor.single_phase_three_phase_switching
          - entity: sensor.charger_enable
          - entity: sensor.wallbox_device_type_code
        title: Raw-Werte
```

katomm, 2024-10-08:
Thanks everybody involved.

hboltz, 2024-10-25:
Thanks for your Dashboard template - I used it but I got 2 error messages:

This dashboard refers to the following entities that are unknown to Home Assistant:

automation.wallbox_charger_remote_control_cmd
input_select.set_wb_remote_control

I checked the [modbus_wallbox.yaml](https://github.com/Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant/blob/main/modbus_wallbox.yaml) and I didn't find these 2 entities. Did you extend the modbus_wallbox.yaml to get them? What am I making wrong?

SunnyCrockett, 2024-10-25:
both entities were removed, I have updated my above dashboard snippet. You can also simple delete both entites from your dashboard.