Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant #12: Experimenting with Charger Card

btota, 2024-02-19:
Currently I'm experimenting with tmjo's card [https://github.com/tmjo/charger-card](url). 
There is a way to use templates [https://github.com/tmjo/charger-card/blob/master/src/templates/const_template.ts](url) to create/add custom brands. For now, I use an existing template which I temporary have adapted to @Louisbertelsmann project. Maybe we can finish the whole thing and get a template ready for Sungrow to use.

[image]
[image]
[image]

```ruby

type: custom:charger-card
entity: sensor.charging_power
brand: sungrow
show_leds: true
details:
  name:
    entity_id: sensor.wallbox_device_type
    attribute: name
  status:
    entity_id: sensor.charging_status
  currentlimits:
    - 6
    - 7
    - 8
    - 9
    - 10
    - 11
    - 12
    - 13
    - 14
    - 15
    - 16
    - 20
    - 32
    - 63
  statetext:
    Available: Available
    Preparing: Plugged in
    SuspendedEV: Stopped (Car)
    Charging: Charging
    Finishing: Finishing
  collapsiblebuttons:
    group1:
      text: Details
      icon: mdi:speedometer
    group2:
      text: Information
      icon: mdi:information
    group3:
      text: Config
      icon: mdi:cog
  info_left:
    - entity_id: sensor.charging_status
      text: Availability
      icon: mdi:gas-station-in-use-outline
  info_right:
    - entity_id: sensor.charging_energy
      text: voltage
      icon: mdi:car-turbocharger
      unit_show: true
    - entity_id: sensor.charging_power
      text: power
      icon: mdi:power-socket-fr
      unit_show: true
  group1:
    - entity_id: sensor.charging_duration
      text: Charging duration
      icon: mdi:timelapse
      unit_show: false
    - entity_id: sensor.charged_range
      text: Charged range
      icon: mdi:highway
      unit_show: false
    - entity_id: sensor.charging_start_time
      text: Last Config Update
      icon: mdi:timer-play
    - entity_id: sensor.charging_end_time
      text: Last Data transfer
      icon: mdi:timer-stop
  group2:
    - entity_id: binary_sensor.charging_status_id
      text: S/N
    - entity_id: sensor.charging_status_id_tag
      text: TagID
    - entity_id: sensor.charging_status_reconnects
      text: Reconnects
    - entity_id: sensor.charging_status_transaction_id
      text: TransactionID
    - entity_id: sensor.charging_status_model
      text: Model
    - entity_id: sensor.charging_status_features
      text: OCPP features
    - entity_id: sensor.charging_status_temperature
      text: Temperature
      unit_show: true
  group3:
    - entity_id: sensor.charging_status_version_firmware
      text: Firmware
    - entity_id: sensor.charging_status_status_firmware
      text: Firmware-status
    - entity_id: sensor.output_current_setting
      text: Current Limit
      icon: mdi:current-ac
      type: dropdown
      service: number.set_value
      service_data:
        entity_id: input_number.set_wb1_output_current
        value: '#SERVICEVAL#'
  stats:
    default:
      - entity_id: sensor.charging_status_stop_reason
        text: Stop reason
      - entity_id: sensor.charging_status_error_code_connector
        text: Error (connector)
      - entity_id: sensor.charging_status_error_code
        text: Error (ocpp)
    Available:
      - entity_id: sensor.charging_status_stop_reason
        text: Stop reason
      - entity_id: sensor.charging_status_error_code_connector
        text: Error (connector)
      - entity_id: sensor.charging_status_error_code
        text: Error (ocpp)
    SuspendedEV:
      - entity_id: sensor.charging_status_energy_session
        text: Energy charged
        unit_show: true
      - entity_id: sensor.charging_status_stop_reason
        text: Stop reason
      - entity_id: sensor.charging_status_error_code_connector
        text: Error (connector)
      - entity_id: sensor.charging_status_error_code
        text: Error (ocpp)
    Preparing:
      - entity_id: sensor.charging_status_stop_reason
        text: Stop reason
    Charging:
      - entity_id: sensor.charging_status_energy_session
        text: session_energy
        unit_show: true
      - entity_id: sensor.charging_status_stop_reason
        text: Stop reason
      - entity_id: sensor.charging_status_error_code_connector
        text: Error (connector)
      - entity_id: sensor.charging_status_error_code
        text: Error (ocpp)
      - entity_id: sensor.charging_status_current_offered
        text: Max Current
      - entity_id: sensor.charging_status_time_session
        text: Charging time elapsed
  toolbar_left:
    default:
      - service: switch.toggle
        service_data:
          entity_id: input_select.set_wb_charger_enable
        text: Toggle on charging
        icon: mdi:play-speed
      - service: button.press
        service_data:
          entity_id: input_select.set_wb_remote_control
        text: Toggle off charger
        icon: hass:stop
      - service: number.set_value
        service_data:
          entity_id: input_number.set_wb1_output_current
          value: '#SERVICEVAL#'
        type: dropdown
        text: Set max current
        icon: hass:current-ac
    Available:
      - service: switch.toggle
        service_data:
          entity_id: undefined
        text: Toggle charging
        icon: hass:flash
      - service: button.press
        service_data:
          entity_id: undefined
        text: Restart charger
        icon: hass:restart
      - service: number.set_value
        service_data:
          entity_id: undefined
          value: '#SERVICEVAL#'
        type: dropdown
        text: Set max current
        icon: hass:current-ac
    SuspendedEV:
      - service: switch.turn_off
        service_data:
          entity_id: undefined
        text: Stop charging
        icon: hass:flash-alert
      - service: button.press
        service_data:
          entity_id: undefined
        text: Restart charger
        icon: hass:restart
      - service: number.set_value
        service_data:
          entity_id: undefined
          value: '#SERVICEVAL#'
        type: dropdown
        text: Set max current
        icon: hass:current-ac
    SuspendedEVSE:
      - service: switch.turn_on
        service_data:
          entity_id: undefined
        text: Stop charing
        icon: hass:flash-alert
      - service: button.press
        service_data:
          entity_id: undefined
        text: Restart charger
        icon: hass:restart
      - service: number.set_value
        service_data:
          entity_id: undefined
          value: '#SERVICEVAL#'
        type: dropdown
        text: Set max current
        icon: hass:current-ac
    Charging:
      - service: switch.turn_off
        service_data:
          entity_id: undefined
        text: Stop charging
        icon: hass:stop
      - service: button.press
        service_data:
          entity_id: undefined
        text: Restart charger
        icon: hass:restart
      - service: number.set_value
        service_data:
          entity_id: undefined
          value: '#SERVICEVAL#'
        type: dropdown
        text: Set max current
        icon: hass:current-ac
    Finishing:
      - service: button.press
        service_data:
          entity_id: undefined
        text: Restart charger
        icon: hass:restart
      - service: switch.toggle
        service_data:
          entity_id: undefined
        text: Toggle charging
        icon: hass:flash
    Preparing:
      - service: switch.turn_on
        service_data:
          entity_id: undefined
        text: Stop charing
        icon: hass:flash-alert
      - service: button.press
        service_data:
          entity_id: undefined
        text: Restart charger
        icon: hass:restart
      - service: number.set_value
        service_data:
          entity_id: undefined
          value: '#SERVICEVAL#'
        type: dropdown
        text: Set max current
        icon: hass:current-ac
chargerImage: White
customImage: \local\sungrow_wb_bl.png
customCardTheme: theme_transp_blue
compact_view: false
show_toolbar: true
show_name: true
show_status: true
show_stats: true
show_collapsibles: true
``

SunnyCrockett, 2024-02-23:
That looks pretty good! I will try to implement it too as soon as I have a bit more time.
Can you tell me what is missing/not working currently?

A bit off topic:
1. I have problems with the charging start and end time, they have an offset of -7 hours. The offset is in HA and also in the iCloudSolar app. Do you know, if it is possible to set the timezone within the wallbox? As an alternative, I could define an offset in the HA config, but thats seems not correct to me.
2. Do ypu know what I have to do, to start the charging? Like the button "Ladevorgang starten" in the iCloudSolar app? I know I can stop the charging with "Remote Contol" -> "Stop charging", but it doesn't with "Start charing"
...I can open seperate issues for that if you like.

btota, 2024-02-26:
1. Yes. You can set systemtime in winet-s
[image]

As I mentioned https://github.com/Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant/issues/6#issuecomment-1937760898 before, within "as_local" I got an offset of +1 hour.

2. I can start/stop via remote control -->input.select{Start/Stop charging}, I shortly tested yesterday on "Lademodus-->PV-Überschuss Laden" in ISC:
bugs: 
#HA/Android: 
##input.select box doesn´t refresh status, like you'll need to manually select and reselect  objects in dropdownlist, maybe this helps you on your start/stop issue.
##Charger enable/disable: enable doen´t work, if selecting disable before --> needs action on ISC/app to re-enable WB.
##Phase switch single/three phase: no action on changing phases
##Set output current: no action on changing ...A
...

Charger Card (in process ...):

works: reading values from modbus
not working: Toggling start/stop, set current output, etc
template https://github.com/Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant/issues/12#issue-2142840880: needs adaption to sungrow

some impressions:

[image]
[image]
[image]
[image]
[image]
[image]

SunnyCrockett, 2024-02-27:
1:
I cross-check my datetime settings in the dongle:
[image]
and also via ModBus cli directly on the wallbox:
[image]
that seems correct for me.
HA is the following:
[image]

2:
I am able to stop charging with the remote control input selector, but not to start it (charging mode in the iCloudSolar app is set to fast charging). I guess I have to drive more often, to have more opportunities to test it ;-)

SunnyCrockett, 2024-02-27:
@btota: Could you upload the image \local\sungrow_wb_bl.png ?

btota, 2024-02-27:
> @btota: Could you upload the image \local\sungrow_wb_bl.png ?

of course! 
[image]

btota, 2024-02-27:
1. it looks good, is there another place in the ISC where the time can be set?
2. I just saw in ISC charging report, that the last charging (maybe more) triggered from HA, wasn´t record?!

SunnyCrockett, 2024-06-04:
You will find my last status of the charger-card here:
https://github.com/SunnyCrockett/Sungrow-SHx-Inverter-Modbus-Home-Assistant/tree/main/dashboards/HACS_Charger