Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant #13: How to change between modes

Eisbaeeer, 2024-03-04:
The isolar app allowes different modes lilke manual, surplus loading, time based loading, etc.

Louisbertelsmann, 2024-03-05:
I think sungrow doesn‘t provide modbus registers for that. I will check later.

pdressen, 2024-03-20:
I have the suspicion, that the "EMS" aka the inverter simply overwrites the modbus registers of the wallbox every few seconds, depending on the charging mode (see #16).

SunnyCrockett, 2024-05-11:
I got some information from Sungrow via ticket regarding this issue. See here:
> The wallbox can only be started/stopped charging via the address 21212, with the charging mode not being stored in the wallbox itself but in the inverter.
> The various charging modes are parameters of the inverter and need to be set from the inverter side.
> 0xA0 fast charging
> 0xA1 green power charging(surplus charging)
> 0xA2 preset charging
> 0xA3 customized charging

And also:
> Please try 33541 charging mode.
> The wallbox can be started/stopped charging via the address 21212 itself.

I could read the register 33541 of the inverter yet, maybe some of you have more luck...?

Eisbaeeer, 2024-05-14:
> I got some information from Sungrow via ticket regarding this issue. See here:
>
>
> And also:
>
>
> I could read the register 33541 of the inverter yet, maybe some of you have more luck...?

That´s it. You safed my day!
You must use the holding register of the Inverter to control the wallbox. Im able to change the modes via modbus. Only the timestamp is not clear. It´s not unix timestamp. Division of 3600 is also wrong. Perhaps anybody has an idea?
The register of modbus is attached.
[Wallbox_holding-hegister-SH10RT_sungrow.txt](https://github.com/Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant/files/15305676/Wallbox_holding-hegister-SH10RT_sungrow.txt)

SunnyCrockett, 2024-05-14:
Thanks for the new registers! How did you get those information? Could you post a screenshot (or something else) of the values you get?
Maybe the timestamps need a "swap: work"-option like some other values of the wallbox/charger?

SunnyCrockett, 2024-05-15:
I have forked the mkaiser yaml for Sungrow inverters and added the registers from @Eisbaeeer. Some of the registers have values, some not. Ich will test a bit when I find time for it. If someone wants to help/test, feel free:
https://github.com/SunnyCrockett/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml

btota, 2024-05-25:
just wrote automation & tested for changing charging_mode (holding reg adress **33540** ??), no effect ... 
changing charging_mode from ISC, changes correct and HA output shows up (160,161,162,163)

 @Eisbaeeer  could you give a feedback?

```ruby

    #Charger wb
  - id: "automation_sungrow_charger_wb_charging_mode"
    alias: "sungrow charger wb charging mode"
    description: "Set charger wb charging mode"
    trigger:
      - platform: state
        entity_id:
          - input_select.set_ac_wb_charging_mode
    condition: []
    variables:
      fast_charging: 0xA0 #160 #fast charging
      surplus_charging: 0xA1 #161 #green power charging(surplus charging)
      preset_charging: 0xA2 #162 #preset charging
      customized_charging: 0xA3 #163 #customized charging
    action:
      - service: modbus.write_register
        data_template:
          hub: SungrowSHx
          slave: !secret sungrow_modbus_slave
          address: 33540 # reg 33541
          value: >
            {% if is_state('input_select.set_ac_wb_charging_mode', "fastCharge") %} 
              {{fast_charging}}
            {% if is_state('input_select.set_ac_wb_charging_mode', "pvSurplus") %} 
              {{surplus_charging}}
            {% if is_state('input_select.set_ac_wb_charging_mode', "quantity&timeBased") %} 
              {{preset_charging}}
            {% if is_state('input_select.set_ac_wb_charging_mode', "userDefined") %} 
              {{customized_charging}}
            {% else %}
              {{surplus_charging}}
            {% endif %}
    mode: single    

  - id: "automation_sungrow_charger_wb_charging_mode_input_select_update"
    alias: "sungrow charger wb_charging mode input select update"
    description: "Updates Sungrow charger wb charging mode input select"
    trigger:
      - platform: state
        entity_id:
          - sensor.charger_wb_charging_mode
    condition:
      - condition: template
        value_template: "{{ not is_state('sensor.charger_wb_charging_mode', 'unavailable') }}"
    action:
      - service: input_select.select_option
        target:
          entity_id: input_select.set_ac_wb_charging_mode
        data:
          option: "{{ states('sensor.charger_wb_charging_mode') }}"
    mode: single

SunnyCrockett, 2024-05-25:
I added a surplus charging with/without grid input to my fork today and it seem to work (I guess I will add an update after write_register later). I will try to help to implement the charging mode next

SunnyCrockett, 2024-05-26:
@btota I got the input for charging mode working for me. Is it working for you, too?
[image]

btota, 2024-05-26:
@SunnyCrockett pretty good job, thx! A short test, beside feedback from ISC - switching between modes works fine (unfortunately my EV was fullcharged, when I saw your post). Charging mode tells me "Unknown charging mode!", because of no charging is started, I think. Looking forward for more testing till next weekend.

SunnyCrockett, 2024-05-26:
Does anyone know how to use modbus.write_register for an uint32 with swap: word? I am trying to update the timestamps but get errors setting i.e. the value 1716764820:

> ERROR (MainThread) [homeassistant.components.automation.set_sungrow_charger_wb_charging_duration] set sungrow charger wb charging duration: Error executing script. Unexpected error for call_service at pos 1: 'H' format requires 0 <= number <= 65535

albertofralbe, 2024-05-27:
[image]
From the docs and the error you posted it seems like you would need to write to the two registers separately

SunnyCrockett, 2024-05-28:
@albertofralbe thanks for the hint!
It works for me now, but you have to switch to the charging mode in prior to changing any depenging values. For example, if you have selected surplus charging, changes of values for preset charging are not written into the inverter.
I also added a template for system datetime and removed the registers which were already in the integration of mkaiser.

Has anyone guessed yet what the unknown (???) registers are for?
[image]

btota, 2024-05-30:
@SunnyCrockett thx, looks pretty good. Can one read the "unknown" values with the "known" regs directly over ISC?

SunnyCrockett, 2024-06-05:
I have removed the unknown registers because they were part of the uint32 values. I found two new ones, one can be used to start/stop charging via the inverter. The other one...I dont know?!
I have uploaded the new versions and also my dashboard for now:
https://github.com/SunnyCrockett/Sungrow-SHx-Inverter-Modbus-Home-Assistant/tree/main/dashboards/HACS_Charger

Ccc-ccccc, 2024-06-22:
@SunnyCrockett Great Job!

I am currently trying to control the wallbox via evcc and modbus via the inverter's Winet-s. While hat works in principal, the inverter's built in EMS interferes with evcc making phase changing via evcc rather impossible.
Did you come across any inverter registers that would allow to disable the inverter's EMS control of the wallbox? So, the inverter would just allow to access the wallbox via modbus...

SunnyCrockett, 2024-06-24:
Thanks!
I dont use evcc, so everything I can say is theoretical. Did you try to change the mode of the wallbox so it works independent from the inverter and then change the registers via the inverter dongle? Or maybe disconnect the wallbox from inverter and connect to the wallbox via WLAN directy? That way evcc could do all the work for you and the inverter doesnt know there is a wallbox.

SunnyCrockett, 2025-02-10:
@cubybox thanks! I hope this will explain them:

- Stop charging by SoC: if enabled, the wallbox will stop charging if the battery level is below the defined value.
- Start charging by SoC: if enabled, the wallbox will start charging if the battery level is above the defined value.

Both are realised by Home Assistant automations. Those are features I wanted to have, I haven't seen them in the app.

cubybox, 2025-02-10:
@SunnyCrockett this is great! I will try it :-)

cubybox, 2025-02-12:
@SunnyCrockett @Louisbertelsmann - 
I found a small error in the modbus_wallbox.yaml. 
Two d's were missing from the input_select.set_wb_charger_enable queries.

[image]

hboltz, 2025-02-12:
Hi guys,

would it possible that on of you (experts) add these registers in the dashboard file?  That would be great!! Thanks ;-)

HoBo

SunnyCrockett, 2025-02-12:
@cubybox: good point! I created a [pull request](https://github.com/Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant/pull/27) for that.

@hboltz: what registers do you mean? If you mean Start/Stop charging by SoC, you will find them in [this dashboard](https://github.com/SunnyCrockett/Sungrow-SHx-Inverter-Modbus-Home-Assistant/tree/main/dashboards/HACS_Charger).

hboltz, 2025-02-14:
@SunnyCrockett : I didn't know this dashboard so far. I followed your setup instructions but I still see some problems:

hboltz, 2025-02-14:
@SunnyCrockett : great job - this is exactly what I was searching for!!!