Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant #16: Control command issues

pdressen, 2024-03-20:
Currently I am only able to stop and restart a running session. I cannot initially start a session via home assistant. The wallbox is in "EMS" mode and connected to an SH8.0RT inverter. Also when I set the output current or the working mode, these settings are reverted after a few seconds. I can confirm, that the commands actually reach the wallbox, because I can see the charging power change briefly and I also double checked using a modbus tcp debug client (EasyModbus). It seems like either the inverter or the iSolarCloud (ocpp?) is overwriting the modbus registers every few seconds.

Did somebody manage to fully control the wallbox over modbus tcp using the Winet-S modbus tcp server?

poehlerflorian, 2024-03-21:
Can confirm this. The output current / charging power is actually changed for a brief moment since the car shows a different estimated charging time. The output current however is then being raised back to 16A.

I disabled internet access for the Winet-S adapter so no data is being sent to or from iSolarCloud. So that probably means either the inverter or the wallbox set it back to 16A automatically.

Louisbertelsmann, 2024-03-22:
I would assume @pdressen is right. If the inverter is set to EMS mode, the inverter would make sure it’s using the inverter commands and not something from outside. You could check that by changing the inverter mode.

poehlerflorian, 2024-03-25:
Are there any requirements to change the working mode? Mine is set to Network and does not seem to change when I try to set it via the input.

Also, what exactly is the difference between the working modes? Does Sungrow provide some documentation for that?

kshala, 2024-05-11:
I'm in a similar situation like @pdressen where holding registers didn't seem to work. neither charging so I investigated. I have an AC011E-01.

My wallbox is set to EMS in the web frontend. idk where else to set it as a customer. Still in HA this yaml showed Network as the work mode. When I use "set work mode" to EMS, HA still shows Network for work mode. I suppose something's wrong in the yaml.

In the Sungrow specs, 21263 "work_mode" is an input register. Not a holding register. I used pymodbus for my own tests but couldn't read from 21263 as input.  Any other holding or input address works fine. So somehtings wrong specifically with 21263.

pymodbus (which is used by HA) doesn't seem to detect an error when reading an address as HOLDING. I assume we believe 21263 is valid as a holding register but actually it's just not working and silently failing? 

My next test would be to change work mode manually to Network and test the rest of the yaml file. Does anybody know why 21263 is set as a holding register in the yaml file?

SunnyCrockett, 2024-05-11:
I changed 21263 from input to holding, because i couldn't get any plausible data as a input. I guess the documentation of Sungrow is not free of errors.
I have requested a newer version via a ticket at Sungrow support, but they told me it is the newest they have.

kshala, 2024-05-11:
@SunnyCrockett reading holding registers never returns an error. It always returns zeros. I've read through all addresses in 21201 to 21400 and every address that should fail returned 0. 

I don't think the spec is wrong. It's the wallbox software that bugs out on 21263 as an input. I don't think work mode is supposed to be writable. I have AC011K-AE-25_V1.4.789 running. How did you reach out to Sungrow?

SunnyCrockett, 2024-05-12:
I am quiet new to Modbus and also yaml/HA and just tried to get it running somehow. If you find something to make the integration better, feel free to do so. 
You can create a ticket here:
https://gsp.sungrow.cn/user/login

kshala, 2024-05-20:
I raised a ticket but didn't get real help reading the mode. However I know for sure the mode is not a holding register. Also I only got data if the mode is set to EMS. Network or Plug and Play made the wallbox not reachable. That said I think the code around 21263 can be removed safely because it can't work.