Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant #24: Integration is not working with my current configuration

BenSki2023, 2024-10-30:
I also have a question about this. Unfortunately, the Louis Bertelsmann integration of [Sungrow-Wallbox-Modbus-HomeAssistant](https://github.com/Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant) is not working for me.

Here's what I've configured or checked:

1. Added [YAML](https://github.com/Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant#212-secretsyaml) entry in "Secrets"

- wallbox_modbus_host_ip: expressif # TODO update with the IP of your inverter's WiNet-S dongle (not ModBus LAN port). No default. Check your router.
- wallbox_modbus_port: 502 # TODO update with the Modbus port of your inverter WiNet-S dongle. Default is '502'.
- wallbox_modbus_slave: 3 # TODO update with the slave address of your wallbox. Default is '3'.

2. Entered the IP address as used for the inverter integration, which works perfectly.

3. Opened the WiNet-S web interface to check the Modbus ID.
[image]

4. Copied [modbus_wallbox.yaml](https://github.com/Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant/blob/main/modbus_wallbox.yaml) to the "integrations" level.
[image]

5. Created a dashboard; I can see many entities, but not all, and 

- some are grayed out or 
- show "Not available."
[image]

Does anyone have a tip for me? 
[image]

[image]
Thanks!

SunnyCrockett, 2024-11-01:
Hi
Are you using the same IP/hostname and port for both integrations? If yes, thats your problem. The ModBus TCP client cannot handle two connections to the same destination.
Try to use the IP and port of the ModBus TCP Port for the mkaiser integration and the IP and port of the dongle (WiNet-S) for the wallbox integration.
If you only want to use the dongle, you have to setup a ModBus TCP proxy with two ports and configure them for your integrations.

PS: if your wallbox/charger is connected to the inverter, check out [my extended mkaiser integration](https://github.com/SunnyCrockett/Sungrow-SHx-Inverter-Modbus-Home-Assistant/tree/main). Together with the integration of Louis Bertelsmann you can have a [dashboard](https://github.com/SunnyCrockett/Sungrow-SHx-Inverter-Modbus-Home-Assistant/tree/main/dashboards/HACS_Charger) with nearly all functions like in the app (its not finisehd completely yet).

Good luck!

SunnyCrockett, 2024-11-06:
Correct, I have forked the mkaiser integration and try to keep it up to date with my charger additions. 

Just to be sure: you need my [sungrow integration](https://github.com/SunnyCrockett/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml) and the [one of Louis Bertelsmann](https://github.com/Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant/blob/main/modbus_wallbox.yaml) to use my [charger dasboard](https://github.com/SunnyCrockett/Sungrow-SHx-Inverter-Modbus-Home-Assistant/tree/main/dashboards/HACS_Charger).

tkone88, 2025-01-18:
Is it possible to Change the modbus_wallbox.yaml so that they use the same IP connection and port for mkaiser and this integration? So i will only have to read the correct slave?

SunnyCrockett, 2025-01-20:
To do so, I guess, you have to copy&paste the entities from the wallbox config to correct sections of the mkaiser config. In theory, that could work but is a of manual work.

Louisbertelsmann, 2025-01-20:
@SunnyCrockett is correct. Home Assistant currently doesn't like two "Hubs" as they call it on the same IP address. It will just spit out an error and not work unless you add those entities directly into the mkaiser yaml file.

tkone88, 2025-01-22:
Did somebody this already?