Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant #25: wb1_dev_number_raw and wb1_working_mode

josch88, 2025-01-12:
Hi,
after installing this integration I got the error in the dashboard that the sensor "wb1_dev_number_raw" is not existing. Maybe the modbus address is not correct?

And the text values of the "wb1_working_mode" seem not to be correct. My wallbox is running in EMS mode but the integration shows "Network". 
Maybe the setup of "Network = 0",  "Plug and Play = 2" and "EMS = 6" is not correct or you read the wrong modbus register?

All the rest of this integration seems to be fine.

SunnyCrockett, 2025-01-15:
Hi josch88

the wb1_dev_number_raw is working fine for me with model AC011E-01. Maybe your wallbox model does not have this modbus register..?

The values of wb1_working_mode are copied from the documentation provided by Sungrow. If they don't fit the values from the iSolarCloud App, feel free to find out the correct values and create a pull request for the repository.

josch88, 2025-01-15:
Hi, i have the same wallbox, so it is strange that the integration tells me on the dashboard that the sensor. wb1_dev_number_raw is not available.

But what about the working mode?

SunnyCrockett, 2025-01-15:
Maybe there is an update available for the wallbox, that will help?

Regarding the working mode:
#
> The values of wb1_working_mode are copied from the documentation provided by Sungrow. If they don't fit the values from the iSolarCloud App, feel free to find out the correct values and create a pull request for the repository.

josch88, 2025-01-15:
The firmware of my wallbox is up to date.

Can you tell my if the working mode you set for your wallbox is fitting to the one shown in home assistant?

SunnyCrockett, 2025-01-16:
I never checked that, and I don't use the app for the inverter/wallbox.

josch88, 2025-01-16:
I guess your Wallbox is running in EMS mode. If this is the case, does your working mode sensor in home assistant say the it runs in EMS?

SunnyCrockett, 2025-01-17:
The value of sensor working mode is "Network" for me, too.

josch88, 2025-01-17:
Ok and is my assumption correct that you configured the wallbox to be in EMS mode?

SunnyCrockett, 2025-01-20:
I don't know what is configured in the wallbox. It was done by the electrician and not me, sorry.

josch88, 2025-01-20:
Ok, if you have a Sungrow inverter and if your wallbox is connect to this inverter, then the working mode is most likely EMS, which means that the working mode number values take from the sungrow docu are incorrect

SunnyCrockett, 2025-01-21:
Sound correct, yes. It would not be the first time the documentation is not 100% correct ;)
If you like, please check out the correct values and open a pull request with the changes.

josch88, 2025-02-02:
Hi, sorry for the late reply. I would love to do this, but on the other side I don't like to mess with the Wallbox for now. 

But maybe you can to me a favor:
I saw you additional charger related changes for the mkaiser sungrow integration in your github account.
Is there any chance that you can integrate this to the wallbox integration instead of the inverter integration?

SunnyCrockett, 2025-02-05:
All additional registers I use in my mkaiser fork are registers of the inverter and not of the wallbox. You have to have an inverter to use them, so I think it makes no sense to put them into the wallbox yaml.

cubybox, 2025-02-12:
> Hi, after installing this integration I got the error in the dashboard that the sensor "wb1_dev_number_raw" is not existing. Maybe the modbus address is not correct?

same here. newest firmware on SH10RT an AC011E. But doesn't really seem relevant to functionality ;-)

[image]

SunnyCrockett, 2025-02-12:
Still works for me, really strange.

If you like, you could try to manually edit the entity in the card. Mine looks like this:
[image]

[image]

josch88, 2025-02-12:
The problem will be fixed with replacing in the yaml file

sensor.wb1_dev_number_raw
with
sensor.wallbox_device_type

cubybox, 2025-02-12:
I haven't even tried searching for another entity - but it works. Thanks! :-)

[image]

hboltz, 2025-02-15:
> Ok, if you have a Sungrow inverter and if your wallbox is connect to this inverter, then the working mode is most likely EMS, which means that the working mode number values take from the sungrow docu are incorrect

I have exactly the same issues - I already replaced sensor.wb1_dev_number_raw with sensor.wallbox_device_type - that's fine now.

Did you find a solution for the "callbox working mode" problem? My wallbox is definitely working in EMS mode, but in HA it shows "network" mode. Are you able to find the right register values?

josch88, 2025-02-15:
Agree with you, but I have no solution and I don’t try to solve it anymore as i will return the Sungrow wallbox and get a Tinkerforge Warp 3

hboltz, 2025-03-03:
anyone here, who found a solution for the "wallbox working mode" problem? My wallbox is definitely working in EMS mode, but in HA it shows "network" mode. Are you able to find the right register values?

SunnyCrockett, 2025-03-05:
@hboltz / @josch88 could one of you do the following:
write down the value of the working mode in the app and the value of working mode in the "Raw" card on the HA dashboard (should be a number). Then change the value in the app and wait till the value in HA changes (takes up to one hour!) and write that down. Repeat till you have done it for every value available in the app and finally post your results here. Then we should be able to fix the integration.