Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant #26: Failed to connect [Errno 111] Connect call failed

hboltz, 2025-01-30:
Hi,

I'm struggling with this "integration" since a few days - before it worked fine. Now I got no connection to the inverter anymore, but the failure:

Logger: pymodbus.logging
Quelle: components/modbus/modbus.py:307
Erstmals aufgetreten: 14:17:16 (1 Vorkommnisse)
Zuletzt protokolliert: 14:17:16

Failed to connect [Errno 111] Connect call failed ('172.17.2.87', 502)

The secrets,vaml is unchanged:

sungrow_modbus_host_ip: 172.17.2.107 
sungrow_modbus_port: 502
sungrow_modbus_slave: 1
wallbox_modbus_host_ip: 172.17.2.87
wallbox_modbus_port: 502 
wallbox_modbus_slave: 2

In the /homeassistant/integrations/modbus_wallbox.yaml, I copied the whole content of the file from this repository.

The sungrow solar app is working fine with the wallbox. I checked the WiNet-S device list - everything is correct like my configuration.

[image]

Any idea of the root cause?

SunnyCrockett, 2025-01-31:
Hi
Did you double-checke the ips on your Router? Maybe the wallbox got a different ip from dhcp.

hboltz, 2025-01-31:
The IPs are correct - the screenshot was made from the WiNet-S Interface, so the IP must be the right one.  But I dubblechecked it once again and can see - this is not my problem. Any other idea?

SunnyCrockett, 2025-02-05:
If you havent done it, try to reboot the wallbox and WiNet-S dongle. Check if updates are available for one or more of your components and apply them.
You could also check if you are able to connect to the IP on port 502 with a telnet client or try to read data with [modbus-cli](https://github.com/favalex/modbus-cli). That way you could assure, that the wallbox/dongle give back data and the IPs are correct.

hboltz, 2025-02-10:
sorry for my delay - I was 1 week on vacation!

To your questions: yes - I rebooted the wallbox and the WiNet-S dongle several times. All devices are up to date:

SH8.0RT-V112 --> 95.01
WiNet-S --> WINET-SV200.001.00.P030
AC011E-01 --> AC011K-AE-25_V1.5.307

I can open the https web gui from the WiNet-S Interface but I cannot telnet to the device on Port 502 - here I get a connection error too. I can telnet to the LAN port on port 502 without any problems. What can be the reason for the problem on WiNet-S on port 502???

hboltz, 2025-02-10:
I found the failure in the configuration of the WiNet-S Interface! But - maybe someone can show me the correct Modbus settings in the WiNet-S GUI?! 

[image]

In my configuration, the 502 switch was switched off! Is SSL encryption needed? What is the port 516 configuration needed for?

SunnyCrockett, 2025-02-10:
Mine looks like this:

[image]

I hope it helps you.

hboltz, 2025-02-10:
Ok - then it‘s okay now! And it works!!! Thanks a lot.

closed 2025-02-10 by hboltz