Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant #31: Wallbox Total Energy is to high

pr0gstar, 2025-05-12:
Hello everyone, and thanks for this project!

I’ve encountered an issue where the sensor `sensor.total_energy` seems too high. For example, in one day it increases by almost 400 kWh, but my car’s battery capacity is only 52 kWh—so that seems incorrect. 🤔

[image]
[image]

Perhaps there’s an error in the calculation for this value, or something else is going on. Has anyone else encountered this issue? It worked correctly in the past.

I’m using the current version of the `modbus_wallbox.yaml` from this repository. If you need any more information about my setup, please let me know. 

Thanks in advance!

SunnyCrockett, 2025-05-13:
Hi
the sensor total_energy is the value of a modbus register of the charger, there is no calulation. I had the same once in the past, the value increased by about 80kwh without using the charger that day.

A few days ago I had "a little" peak, but it decreased again with the next update:
[image]

Maybe it has something to do with the firmware of the charger...? It is no help, but you are not alone :)

Cheers!

pr0gstar, 2026-03-15:
Still problems with total energy. It increases by over 200kwh today while charging mit 50kwh electric vehicle 🤦